orion-project/rezonator2 #2: Wrong RTM calculation

mudzin, 2020-03-05:
I do have following scheme with calculated RTM as shown in screenshot.    
[image]

But, if I calculate it manually (both python and by hand) I get 
```
RTM = [ [ 0.6021979   0.19890113]
        [-1.999999    1.0000004 ]
]
```
which is different. I use ABCD (given by reZonator) for each individual component and get same result.

orion-project, 2020-03-06:
Can you please attach the schema file causing the problem and the python code you used for checking?

mudzin, 2020-03-06:
The difference I see here is my ABCD refraction matrix from medium to air on convex curvature (center located after interface - back propagation).

`M_s1_f_t = np.matrix([[1, 0], [8.2e-7, 1.82]])` - in my case.
[image]
```
initial n1 = 1.82
final n2 = 1
R = 1e6

A = 1
B = 0
C = (1.82-1) / 1*1e6 = 8.2e-7
D = 1.82 / 1 = 1.82
``` 
Maybe I messed up something, but I believe ABCD calculation is fine.

orion-project, 2020-03-07:
Lets considert it step-by-step:

0) How we calculate the round-trip regarding to `d2`:
`d2 * s1 (forward propagation) * d1 * M1 * d1 * s1 (back propagation) * d2 * F1`

1) `s1` Forward propagation - the beam *enters* the medium, `n1=1`, `n2=1.82`, surface is concave `R<0`.
`C = (n1-n2)/(R*n2) = (1-1.82)/(-1e6*1.82) = 4.50549e-07`
reZonator uses a bit different convention and to get a concave surface you have to set `R>0`, but it also swaps `n1` and `n2` in formula for `C` and eventually it gives the same result:
`C = (n2-n1)/(R*n2) = (1.82-1)/(1e6*1.82) = 4.50549e-07`

2) `s1` Back propagation - the beam *exits* the medium, `n1=1.82`, `n2=1`, surface is convex `R>0`:
`C = (n1-n2)/(R*n2) = (1.82-1)/(1e6*1) =  8.2e-07`
Again, reZonator uses a bit different convention and to get a convex surface you have to set `R<0`, but it also swaps `n1` and `n2` in formula for `C` and eventually it gives the same result:
`C = (n2-n1)/(R*n2) = (1-1.82)/(-1e6) = 8.2e-07`

Both of these values you can see in the matrix window for `s1`.

mudzin, 2020-03-10:
Ad. 0 - calculation sequence
d2 * s1 (forward) - here, after passing d2 in medium, beam is **_leaving_** it through convex surface (from beam's point of view), so I believe rezonator should put (Ad.2) s1 **back-propagating matrix (C=8.2e-7)**, not _forward_. 

Ad. 1,2 - You are correct and I agree - no doubt.

orion-project, 2020-03-11:
The common rule of matrix optics is that element matrices are multiplied in the direction *opposite* to the direction of the beam propagation. Here it is mentioned in the doc http://rezonator.orion-project.org/help/round_trip_matrix.html. So when you calculate round-trip matrix regarding to d2, you say "the beam traverced d2, and before that it *entered* d2 from air, and before that it traverced d1, and before that it reflected M1" and so on.

orion-project, 2020-03-20:
Since there is no activity, I think it's all clear now so close the issue. Please reopen if there are additional questions,

closed 2020-03-20 by orion-project