orion-project/rezonator2 #4: Resonator mirror assuming vacuum environment?

janfiser, 2021-04-06:
Hello,

I think I found a bug: beam size at resonator mirrors is calculated as if mirrors are surrounded by vacuum even though I specified that they are surrounded by a medium with refractive index n. 

Please see the attached screenshot. My resonator is very simple, it consists of a laser rod with mirrors coated on its end surfaces. When beam size at rod's end surfaces is calculated, it returns the correct result (wavelength inside the rod material is used in the formula). When beam size is calculated at mirror surface, it returns wrong result (wavelength in vacuum is used in the formula even though mirrors are "immersed" in rod material). Beam size at rod surface and at mirror surface should match, because this is the same surface. But in the attached example, they do not match, beam size is discontinuous.

If I somehow set-up my schema wrong, please correct me and excuse me for false alarm.

Best regards.

[image]

Chunosov, 2021-04-13:
Hello @janfiser, thanks for catching. Yes, mirrors always were supposed to be in air. But you schema makes total sense so should be calculated correctly. This is fixed in https://github.com/orion-project/rezonator2/releases/tag/v2.0.8-beta4. Please check.

janfiser, 2021-04-13:
Hello @Chunosov, I tested v2.0.8-beta4 and it works. Thank you for a fast bug fix and keep on the good work with this awesome software. 
[image]

closed 2021-04-13 by Chunosov