orion-project/rezonator2 #17: Export of all caustic data

photoniker, 2022-01-31:
Is it possible to export the whole caustic data for further calculations like it is possible of segments of the caustif between optical elements. 
[image]

orion-project, 2022-01-31:
Currently not, unfortunately. Will add to the roadmap.

photoniker, 2022-02-01:
Can you also add to change the axis labels of the plot. I'm using the plot often with diameter data. In the version 2 it is not possible to change the y label (beam radius).

orion-project, 2022-02-01:
@photoniker Yes, the axis renaming is in the list. But can you elaborate a bit on what you mean "using the plot often with diameter data"? rezonator doesn't calculate diameteres, they are always radii.

photoniker, 2022-02-01:
When you set the w0 as a d4sigma diameter and multiple the M² by 4 you see the caustic plot for the diameter.

bichdong10, 2022-02-09:
I calculate the beam radius for our laser cavity. I wonder about the beam radius is the same with the real setup or not. Does it depend on the diameter of laser rod or the focal length of mirror?
Thanks

Chunosov, 2022-02-09:
@bichdong10 beam radius doesn't depend on the laser rod diameter (rezonator doesn't calculate it though), but it definitely depends on mirrors focuses (and this can be calculated with this app)

Chunosov, 2022-02-10:
Aavailable in 2.0.10-beta6
- commands for changing axes titles for function plots.
- command for exporting all graph segments

closed 2022-02-10 by Chunosov