orion-project/rezonator2 #25: Simulating cavity with incoupling optics

grychtol, 2022-11-21:
Dear all,

I was wondering whether it would be possible to simulate a multipass heriot cell consisting of two curved focusing mirror, while the laser beam is coupled into this cavity via a telescope which is used to adjust the divergence and mode of the incoupling beam. So basically I have divergent laser beam, which is focused via a two lens telescope into a cavity comprising two curved mirrors.

Any idea whether this possible and if so, on how to do so?

Cheers,
Patrik

orion-project, 2022-11-22:
Hello Patric @grychtol 

It seems, a cell is just a sequence of spherical mirrors put on a distance d = 2*R one from another. So you can model it like that.

Using the "Parameters" feature you can easily adjust the whole cell without a need to edit each mirror and distance between them. Here I defined two parameters `cell_R` and `cell_D = 2*cell_R`. (Values are arbitrary, just for illustration the idea). Then I linked radii of M1, M2, M3 to the `cell_R` parameter, and distances d4, d5 are linked to the `cell_D` parameter. Change the `cell_D` parameter value in the Parameter window and all elements and plots will be recalculated. Example schema attached.

[image]

[heriot_cell.zip](https://github.com/orion-project/rezonator2/files/10064439/heriot_cell.zip)

orion-project, 2022-11-22:
> distance d = 2*R one from another

Sorry, this is not quite right. The distance can be arbitrary of course. But the idea is the same.

Some manual work is required still - knowing the distance and the entrance angle you have to calculate a number of passes.

grychtol, 2022-11-22:
Thanks a lot!

This already helped a lot!

I have now updated the schema you kindly provided with more specific parameters of my optical setup and I was wondering whether it would be somehow possible to systematically optimize the position of the telescope and the distance of the lenses in order to equalize the spot sizes in the cavity, i.e. basically that the radius after each round trip is more or less the same?

[heriot_cell_v2.zip](https://github.com/orion-project/rezonator2/files/10065900/heriot_cell_v2.zip)

orion-project, 2022-11-22:
Unfortunately, the app doesn't provide a lot of automation for such optimizations. It's tries to be 'a generic tool', while most of optimization tasks are very particular.

For optimization of the telescope position and its length, you might find useful the "Beamsize Variation" function. Here you can smoothly change one of the element parameter and look how the beamsize  changes somewhere in the system (e.g. at the cell exit).

[image]

orion-project, 2022-11-22:
Another option could be the "Adjustment" tool, it's quite manual though, but still.
Here you can select a parameter and click Plus and Minus buttons and see how the plot changes.

[image]

grychtol, 2022-11-22:
Thanks a lot, both tools seem to be helpful ... however I do not manage to get the adjustment of tele_D working ... add the first adjuster as shown in your screen shot, but I cannot add the one underneath to manipulate tele_D

grychtol, 2022-11-22:
Now, I figured it out!

orion-project, 2022-11-22:
Yes, you don't need to use the first one indeed, because when an element linked to a parameter it can not be changed directly. Only the linked parameter should be changed.

closed 2022-11-27 by orion-project