orion-project/rezonator2 #30: Matrix product

photoniker, 2023-01-16:
Hi, 

when calculating the forward/ backward matrix of some elemente, I see a not expected behaviour:
For the same elements I would expect the inverse matrix: 
[image]

That is not the case for the sign of the values as you can see here: 
[image]

photoniker, 2023-01-16:
ah I found that this is all fine when calculating the matrix. When you calculate the propagation forward and than backwards you have to inverse the matrix.

photoniker, 2023-01-16:
Another question pop out for the definition of forward and backward.
[image]

In the ABCD formalism you calculate the forward propagation matrix by starting the matrix multiplications from the last element. That is in rezonator the other way round. What's the reason?

Chunosov, 2023-01-16:
> That is in rezonator the other way round. What's the reason?

'Forward' in 'backward' are not about 'beam propagation' in these commands. They just say 'multiply from the first to last selected' (forward), and 'multiply from the last to first selected' (backward). That's all, sorry for the confusion. I think it's worth renaming them to something more clean. They are rather debug command used to compare multiplication result with what the test Python code gives.

>  I found that this is all fine when calculating the matrix

Yes, rezonator doesn't do analytical inversion, it just multiplies matrices one-by-one following the light path. The below test code shows that the multiplication is correct. We can see that back propagation matrix is not exactly an inverted forward one:

```python
import numpy as np
from numpy.linalg import inv

M_L1_t = np.matrix([[1, 0.1], [0, 1]])
M_M1_t = np.matrix([[1, 0], [-20, 1]])
M_L2_t = np.matrix([[1, 0.05], [0, 1]])
M_M2_t = np.matrix([[1, 0], [-13.33333333333333, 1]])

M0_t = M_L1_t * M_M1_t * M_L2_t * M_M2_t
M0_t_1 = M_M2_t * M_L2_t * M_M1_t * M_L1_t
M0_t_inv = inv(M0_t)
print('M0_t\n', M0_t, '\n')
print('M0_t_1\n', M0_t_1, '\n')
print('M0_t_inv\n', M0_t_inv, '\n')    
```

```
M0_t
 [[ -1.66666667   0.05      ]
 [-20.           0.        ]] 

M0_t_1
 [[  0.           0.05      ]
 [-20.          -1.66666667]] 

M0_t_inv
 [[ 0.         -0.05      ]
 [20.         -1.66666667]] 
```

photoniker, 2023-01-17:
Yes, it renaming would help to make it clear ;-)

orion-project, 2023-05-03:
[v2.0.12-beta8](https://github.com/orion-project/rezonator2/releases/tag/v2.0.12-beta8)

closed 2023-05-03 by orion-project