orion-project/rezonator2 #31: Gaussian apertures don't seem to work in RR or SW modes

Na-Cleric, 2023-02-10:
Hi, i've been experinmenting with the gaussian aperture and gaussian aperture with lens elements, and i've noticed that reZonator returns nan values for beam parameters in Ring resonator and Linear resonator modes whenever the a2 value is nonzero (the ABCD matrix is complex). It still shows that the resonator is stable. The elements seem to work ok in SP mode.
[image]
[image]

Chunosov, 2023-02-15:
Hello @Na-Cleric 

Yes, elements with complex matrices were added mostly for SP systems. Because it's all very clear for SP systems, complex beam parameter is transformed by complex matrices and radius and ROC can be [calculated definitely](http://rezonator.orion-project.org/help/gauss.html).

Properly, it should be some warning against using such elements in resonators, but currently there is no way to restrict user to use any elements in any system.

As for resonators, I'm even not sure if simplified [ABCD to w,R](http://rezonator.orion-project.org/help/matrix_optics.html) equations are valid for complex round-trip matrices.

I'll check what the program calculates for complex round-trips, but right now I'd not trust to that green stable marker :)

orion-project, 2023-05-03:
[v2.0.12-beta8](https://github.com/orion-project/rezonator2/releases/tag/v2.0.12-beta8)

closed 2023-05-03 by orion-project