orion-project/rezonator2 #41: Feature request: Add z0, d0, M² to "Beam Parameters at Elements"

oqb, 2023-09-26:
First off: thanks for this awesome tool!

Very often, I would like to find the beam waist diameter, beam waist distance and M² value of a laser beam behind a lens and maybe some further optics. In Rezonator, I would model this as a single-pass system and plot the multi-range caustic and manually read off the parameters. Alternatively, I can use the "Beam Parameters at Elements" feature, which returns the Ws, Rs, and Vs parameters, from which I can with some math derive w0, z0, and M². 

I would highly appreciate if rezonator2 would display these 3 parameters directly in the Beam Data table.

oqb, 2023-11-08:
Dear Chunosov,
thanks for your reply. Having the caustic parameters in the caustic window sounds great and serves my purposes.

Concerning M²: No need to include it into the caustic window. I have the triple (w0,z0,M²) hard-wired in my mind, that's why I listed it above. You are absolutely right, it shouldn't change in rezonator propagation.

orion-project, 2023-11-28:
Available in https://github.com/orion-project/rezonator2/releases/tag/v2.0.13-beta9

closed 2023-11-28 by orion-project