orion-project/rezonator2 #42: Beam size exhibits discontinuous jump at "tilted plane-parallel slab" boundaries

janfiser, 2023-11-03:
Hello,

the simple schema below yields a strange result. Beam size exhibits discontinuous jump at "tilted plane-parallel slab" boundaries. This happens only for T-polarization. Discontinuity disappears if tilt is set to 0° or if refractive index is set to 1, so it seems it is connected to refraction.

If I overlooked some basic physics, please excuse me for submitting a non-existing issue.

Best regards.

[image]

orion-project, 2023-11-04:
Hello @janfiser 

This is an effect of how the beams size is measured. Please look at the figure:

[image]

Beamsize is measured perpendicular to its axis, not along the element edge. So while the beam is obviously continuous on the edge, its projections onto the axis' perpendicular have different sizes before and after the edge.

janfiser, 2023-11-04:
Hello @orion-project,

thank you for a fast reply. I was actually considering this option before submitting the issue, but discarded it since it seems to me that the same discontinuity should happen also in the S-plane because of beam's circularly symmetric cross section?

Best regards.

orion-project, 2023-11-04:
This is essential difference between planes - when we set a titling angle, this means the beam falls at this angle in the T-plane. While from the other side (in the S-plane) the beam falls normally

[image]

If we look at the [S-matrix for tilted plate](http://rezonator.orion-project.org/help/matrix/ElemTiltedCrystal.html):

[image]

it looks the same as matrix for [normal plate](http://rezonator.orion-project.org/help/matrix/ElemPlate.html):

[image]

janfiser, 2023-11-04:
Aha, thank you for an explanation. I think I understand now.
[image]

Just a side note, that isn't related to the issue. In my opinion it does not make much physical sense to talk about beam size in the region where one part of the beam cross section already starts to refract, while the other does not (the region between two violet lines in the picture below). I cannot think of a meaningful/physical definition in this region? But I agree that the definition used in reZonator2 is totally valid in the region where beam size can be meaningfully defined (before first and after second violet line) and that beam size indeed exhibits discontinuous jump.
[image]

Best regards.

orion-project, 2023-11-05:
Yes, you right, in the point of refraction this "beamsize" is not something quite "physical". But the whole matrix approach and the gaussian beam itself are only approximations for real beams. And partially due to the reason you mentioned this approximation is only valid for small tilting angles where `angle ~ sin(angle)`.

If we take for example 100mkm beam and 10deg tilting angle, then the "exact" beamsize on the edge will be 101.5mkm. And nor of these sizes will precise reflect physical reality. So I don't think it makes sence itroducing additional calculation complexity for accounting this mismatch. At least I don't remember I have met this in "classical" books. They always say about "beam size at the edge", meaning the point of refraction and measuring the beamsize perpendicular to the axis.

But yes, we should always have in mind that the larger the beam and the greater the titling angle the less accuracy the abcd-approach gives.

janfiser, 2023-11-05:
I totally agree with you and I think everything was clarified. Thank you for the discussion.
Please keep on with the good work on this awesome software!
Best regards.

closed 2023-11-06 by orion-project