orion-project/rezonator2 #54: Beam size variation results change unexpectedly

farbefreak, 2024-04-09:
Hello!

I think I found a bug.

If you make a simple resonator, add a thermal lens and try to simulate its effect on beam size, something goes wrong.

Starting with a simple resonator and plotting beam size at the thermal lens:
[image]

So far looks ok. 
However if we now change the original input of the focal length, I would expect that the beam size variation plot stays the same.

The MR caustic changes as expected but the beam size variation plot at the TL shouldn't change but it does.
[image]

Looks like ReZonator somehow adds the original input and the plotting variable together? 

(Thanks for the great software BTW!)

orion-project, 2024-05-28:
The fix will be available in 2.0.14 (comming soon, reopening while it's not ready)

orion-project, 2024-06-09:
https://github.com/orion-project/rezonator2/releases/tag/v2.0.14-rc1

closed 2024-06-09 by orion-project

orion-project, 2024-06-10:
@farbefreak should be fixed now in 2.0.14. You were right, it used kind of both - statically assigned and varying values in a weird way.

Thanks for catching! and sorry for it took so long.

PR #57 (2024-05-28): Wrong Beam Variation when plotting in the same range element as being changed
Fixes #54