orion-project/rezonator2 #55: UI problem under macOS dark theme (black text on the dark background)

zhiyuanzhai, 2024-04-28:
[image]

There are some problems with dark UI in macOS, the most notable issue being that some elements and texts remain black on the dark background.

(BTW, sorry if this is a dumb question, but is there a way to turn off the dark scheme for this specific app to avoid problems with the dark UI?)

Chunosov, 2024-05-05:
rezonator dosen't properly support dark themes yet, this is need to be added.

>  is there a way to turn off the dark scheme for this specific app to avoid problems with the dark UI?

Yes, it is thankfully - open a terminal and type

```bash
defaults write com.yourcompany.rezonator NSRequiresAquaSystemAppearance -bool Yes
```
then the app should always be in light theme.

`com.yourcompany.rezonator` is an application bundle name (which will be changed to `org.orion-project.rezontaor`, so you'll probably need to apply the fix again after next update)

This way you can "fix" other apps in your system. To find app bundle name use the command

```bash
osascript -e 'id of app "NAME_OF_APP"'
```

for example 

```bash
osascript -e 'id of app "safari"'
> com.apple.Safari
```

orion-project, 2024-06-09:
https://github.com/orion-project/rezonator2/releases/tag/v2.0.14-rc1

closed 2024-06-09 by orion-project

orion-project, 2024-06-10:
@zhiyuanzhai please check the new version, it should have same colors on both light and dark themes

And thanks for catching! I don't use the dark theme so haven't noticed the issue, but I know people like dark themes so the app should be usable on both. Unfortunately I don't have time to support both light and dark theme so the app will be light on both.