orion-project/rezonator2 #64: Schema layout: infinite ROC

janfiser, 2024-09-12:
Hello,

Lensmaker tool has one really nice functionality: the user can make a plane surface by inputting zero radius of curvature for the surface. This is also reflected in the drawing of the lens where a plane surface is shown. Would it be possible to include this functionality also in Schema Layout? 

Schema: correct me if I am wrong, but at the moment the user cannot input zero radius of curvature (notification "Curvature radius can not be zero.") and must resort to inputting a really high number if he wants an effectively plane surface. Also, the sketch of the lens in Layout is a bit misleading, because it shows a curved surface.

Thank you very much for developing this awesome software!
Best regards.

[image]

orion-project, 2024-09-23:
Hello @janfiser,

Yes, that could be very usefull indeed, thanks for the suggestion!
Added this to the feature list for the next version.

(Sorry for the delated responce)

orion-project, 2025-06-17:
[v2.0.15-rc2](https://github.com/orion-project/rezonator2/releases/tag/v2.0.15-rc2)

closed 2025-06-17 by orion-project