orion-project/rezonator2 #77: Beam Data: Point element representation

janfiser, 2025-07-02:
Hello,

I noticed that a Point element in Beam Data is depicted as a lens with "before/after properties" if it is not surrounded on both sides with other elements (e.g. free space). Is this by design and I am missing something?

Thank you for developing the software!
Best regards.

[image]

orion-project, 2025-07-07:
Hi @janfiser,
It's not intentional, it's rather a side effect of calculation algorithm. 
For this function all elements are splitted into several kinds
 
- "simple" elements for wich only parameters "at" are calculated
- "wavefront changing" elements for wich parameters "before" and "after" are calculated
- "empty" elements wich only splits the previou kinds and not calculated by default

The point is a "simple" element so beam data only calculated at the point as on the right screenshot. This is its default representation.
Another thing is the input beam values at the schema entering - they also marked as "before" for the first element. This is what you see on the left screenshot.

This was one of the Point's purpose - to put it where you want to know the data, including at the schema start. Though I agree - "before" and "after" values spoils this purpose a bit. I'll check, maybe it worth adding another condition and show points simpler.

Another option - you can click the Options button (that one with checkmarks on the toolbar) and select "Calculate in empty spaces". This will show values at the beginning and the end of d16 on the right screenshot. This way one also could see the beam at the schema entering. But it also shows many unnecessary values in other places. So it's more like a debug option.

orion-project, 2025-08-11:
Changed in [v2.1](https://github.com/orion-project/rezonator2/releases/tag/v2.1.0).

closed 2025-08-11 by orion-project